nerdyscout/kicad-exports #3: BOM generation errors

brainstorm, 2020-06-06:
This project is awesome @nerdyscout, kudos! :)

The `ibom.html` got generated correctly, but the `.csv` file only contains the header:

```
Reference, Quantity, Value, Footprint, Datasheet
```

See the full backtrace while running kicad-exports.sh locally:

```
$ ../kicad-exports/kicad-exports.sh bom bom
running OUTSIDE docker...
running INSIDE docker...
DIR=bom
SCHEMATIC=example_board.sch
BOARD=example_board.kicad_pcb
PROJECT=example_board.pro
NAME=example_board
MANUFACTURER=
PARAMETERS=
VERBOSE=-v
INFO:Waiting for "Main eeschema window" ...
INFO:Open Tools->Generate Bill of Materials
INFO:Waiting for "Bill of Material dialog" ...
INFO:Paste xslt command
INFO:Generating netlist
INFO:Wait for BoM file creation
INFO:Closing dialog
INFO:Waiting for "Main eeschema window" ...
INFO:Exiting eeschema
rm: cannot remove 'bom/example_board.xml': No such file or directory
Git process cannot be launched: [Errno 2] No such file or directory: 'git': 'git'
Loading example_board.kicad_pcb
2020-06-06 12:40:18,720 INFO Unsupported drawing class PCB_TARGET, skipping
2020-06-06 12:40:19,348 INFO Compressing pcb data
2020-06-06 12:40:19,894 INFO Dumping pcb data
2020-06-06 12:40:19,917 INFO Created file /mnt/bom/ibom.html
INFO:Waiting for "Main eeschema window" ...
INFO:Waiting for "Main eeschema window" ...
ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:132)
Traceback (most recent call last):
  File "/usr/local/bin/eeschema_do", line 536, in <module>
    eeschema_bom_xml_commands(output_file, eeschema_proc.pid)
  File "/usr/local/bin/eeschema_do", line 314, in eeschema_bom_xml_commands
    wait_eeschema_start()
  File "/usr/local/bin/eeschema_do", line 130, in wait_eeschema_start
    wait_eeschema(5)
  File "/usr/local/bin/eeschema_do", line 99, in wait_eeschema
    return wait_for_window('Main eeschema window', r'Eeschema.*\.sch', time, others=others)
  File "/usr/local/lib/python3.7/dist-packages/kicad_auto/ui_automation.py", line 303, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Main eeschema window window
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.7/dist-packages/kicost/__main__.py", line 320, in <module>
    main()
  File "/usr/local/lib/python3.7/dist-packages/kicost/__main__.py", line 311, in main
    dist_list=dist_list, currency=args.currency)
  File "/usr/local/lib/python3.7/dist-packages/kicost/kicost.py", line 164, in kicost
    p, info = eda_module.get_part_groups(in_file[i_prj], ignore_fields, variant[i_prj])
  File "/usr/local/lib/python3.7/dist-packages/kicost/edas/eda_kicad.py", line 110, in get_part_groups
    file_h = open(in_file)
FileNotFoundError: [Errno 2] No such file or directory: 'bom/example_board.xml'
rm: cannot remove 'bom/example_board.xml': No such file or directory
```

brainstorm, 2020-06-06:
Ha! Now I get why it doesn't work, my board is generated with [SkiDL](https://xesscorp.github.io/skidl/docs/_site/) since it's just a very repetitive placement of resistors, so there's no schema at all, only PCB layout... no wonder whatever it tries to do with `eeschema` fails since it's empty :)

nerdyscout, 2020-06-07:
honestly I am not sure how to handle this issue... kicad-exports was not meant to be used without kicad files (*.sch, *.kicad_pcb). SkiDL is new to me, never heard of it yet, but it looks very interesting.
One lesson I learned is about which parameters in the actions.yml are optional, I might change this with the next version... but it does not help you running it manually.
PCBNew is capable of generating BOMs, but the used kicad-automation-scripts is currently not. I would be happy to integrate it, but have either the knowledge nor the time.

brainstorm, 2020-06-08:
Gotcha @nerdyscout... thing is that while SKiDL bypasses the need for a `.sch` file, `.kicad_pcb` is still there, so this `bom` function successfully generates an `ibom.html` (which contains all the useful information) but crashes as shown above looking for the `.xml` file.

I hear you and don't have the time either, let's leave this open just in case someone stumbles with the same issue?

Thanks!

nerdyscout, 2020-06-08:
yes, I will leave it open.
next version will contain some more documentation to point this problem out.
I suggest you to use `../kicad-exports/kicad-exports.sh ibom bom` which should create your ibom.html without triggering the failing stuff.

nerdyscout, 2020-08-08:
hi @brainstorm 
there is a v2.0 pre-release now with major changes. I would like to ask you to try this one as I expect to solve your issue here.

nerdyscout, 2020-08-21:
closing as this should be no issue anymore with the released v2.0

closed 2020-08-21 by nerdyscout