nerdyscout/kicad-exports #6: Generate combined board PDF of all layers

dzervas, 2020-07-30:
Currently command `kicad-board` generates 2 PDFs - one with the front and one with the back. I'd really like to either alter or add a command to generate a PDF with both of the layers (or all of the layers in case of multi layer board)

nerdyscout, 2020-07-30:
I am more than happy to except any change request, please make them based on the development branch.
You might notice there is already a option for a combined pdf if a specific output file is given.

bad news: I just had an idea of a complete new approach for this project... so it might be to not develop kicad-exports anymore further.

nerdyscout, 2020-08-01:
I just made an update to the development branch. it does include several breaking changes... 
Now, instead of passing/selecting a command, a config file is needed. this file will specify the type of output. Some example files will be part of kicad-exports, but you could specify your own as well, therefore this is much more flexible.
So fare I didn't try to generate a multi page pdf, one pdf per layer works.

joaoantoniocardoso, 2020-08-02:
I think it could be done independently of kicad-exports by running a ghost script to combine all pdfs right after kicad-exports, maybe something like `gs -dNOPAUSE -sDEVICE=pdfwrite -sOUTPUTFILE=AllCombined.pdf -dBATCH *.pdf`.

nerdyscout, 2020-08-08:
using pre-release v2.0 i guess you could try some config like this
```
kiplot:
  version: 1

outputs:
  - name: PDF
    comment: "PDF files"
    type: pdf
    dir: PDF
    options:
      exclude_edge_layer: false
      exclude_pads_from_silkscreen: false
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true

      # PDF options
      drill_marks: small
      mirror_plot: false
      negative_plot: false
      line_width: 0.01
    layers:
# choose whichever you need
      - all 
```
https://github.com/INTI-CMNB/kiplot#the-configuration-file

and merge the resulting pdfs afterwards like suggested.

nerdyscout, 2020-08-21:
closing as this should be no issue anymore with the released v2.0

closed 2020-08-21 by nerdyscout