nerdyscout/kicad-exports #7: Use https instead of SSH for submodules

joaoantoniocardoso, 2020-08-02:
Hi maintainers, my first interaction here, so first let me thank you for creating this project. I want to use it in all the projects from [Zenite Solar](https://github.com/ZeniteSolar), then I am offering my efforts to make it better.

---

### Use https instead of SSH for submodules

If I run `git clone --recursive https://github.com/nerdyscout/kicad-exports` it fails:
```
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.
```

I noticed that it is using SSH for submodules, then I manually changed `.gitmodules` to use https and it started to work:

```diff
--- a/.gitmodules
+++ b/.gitmodules
@@ -1,21 +1,21 @@
 [submodule "submodules/KiBoM"]
        path = submodules/KiBoM
-       url = git@github.com:INTI-CMNB/KiBoM.git
+       url = https://github.com/INTI-CMNB/KiBoM.git
 [submodule "submodules/kicad-git-filters"]
        path = submodules/kicad-git-filters
-       url = git@github.com:INTI-CMNB/kicad-git-filters.git
+       url = https://github.com/INTI-CMNB/kicad-git-filters.git
 [submodule "submodules/PcbDraw-Lib"]
        path = submodules/PcbDraw-Lib
-       url = git@github.com:INTI-CMNB/PcbDraw-Lib.git
+       url = https://github.com/INTI-CMNB/PcbDraw-Lib.git
 [submodule "submodules/PcbDraw"]
        path = submodules/PcbDraw
-       url = git@github.com:INTI-CMNB/PcbDraw.git
+       url = https://github.com/INTI-CMNB/PcbDraw.git
 [submodule "submodules/kiplot"]
        path = submodules/kiplot
        url = https://github.com/nerdyscout/kiplot.git
 [submodule "submodules/kicad-automation-scripts"]
        path = submodules/kicad-automation-scripts
-       url = git@github.com:INTI-CMNB/kicad-automation-scripts.git
+       url = https://github.com/INTI-CMNB/kicad-automation-scripts.git
 [submodule "submodules/InteractiveHtmlBom"]
        path = submodules/InteractiveHtmlBom
-       url = git@github.com:INTI-CMNB/InteractiveHtmlBom.git
+       url = https://github.com/INTI-CMNB/InteractiveHtmlBom.git

```

I don't know if it could result in some side-effect on your development but possibly https is a more likely to broadly work.

Thank you

nerdyscout, 2020-08-02:
yes I agree. but somehow I f*cked up the submodules. when i change everything to https it does not work anymore.

joaoantoniocardoso, 2020-08-02:
Hmm, try to look at your `.git/config`, possibly you need to manually change it there too.

Also, wipe out your submodules folder `\rm- rf submodule`, than init `git submodule init` and update `git submodule update` them.

nerdyscout, 2020-08-03:
should be fixed now.
thank you!

closed 2020-08-03 by joaoantoniocardoso