nerdyscout/kicad-exports #8: timeout from eeschema_do

joaoantoniocardoso, 2020-08-02:
Hi, I am trying to use v2.0 but I am getting a timetout error from eeschema_do:
```
ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:137)
```
[see traceback here](https://github.com/ZeniteSolar/MCB19/runs/937996810?check_suite_focus=true#step:4:13)

I am not sure if I am not doing something wrong related to the usage of this github action, any suggestion on how to proceed?

nerdyscout, 2020-08-02:
sry v2 is currently not meant to be used... I am still trying to fix some (other) issues which i don't understand so far.
but thank you for this information, this is new to me.
if you want to digg deeper into I would suggest you to run it locally and add a '-v' to the line calling kiplot in entrypoint.sh

joaoantoniocardoso, 2020-08-02:
Thanks for the information, I will take a look =)

nerdyscout, 2020-08-06:
hi @joaoantoniocardoso thank you a lot for all of this contributions!
is this issue here still a problem?

the test project is running now on the development branch, before merging to master I will run some other projects though it to be sure the output is as expected. but i am very optimistic :smile:

joaoantoniocardoso, 2020-08-06:
Oh, nice to hear that, I am kinda proud to be able to contribute to such a helpful project, sir. Probably I will be using this a lot in near future.

I have been following your advancements this week and I am very optimistic too, I will try to make my project to run this development version right now and as soon as I get something I'll post the feedback here.

joaoantoniocardoso, 2020-08-07:
Hi @nerdyscout. I did the test and it is almost there.
To be able to use `uses: nerdyscout/kicad-exports@develop` in my workflow, It seems that we'll need to update `action.yaml` to use an already compiled, updated docker image:
```
  Step 6/23 : COPY submodules/InteractiveHtmlBom/InteractiveHtmlBom /opt/InteractiveHtmlBom/
  COPY failed: stat /var/lib/docker/tmp/docker-builder322038178/submodules/InteractiveHtmlBom/InteractiveHtmlBom: no such file or directory

[error]Docker build failed with exit code 1
``` 

To test this hypothesis I managed to edit my kicad-exports fork accordingly (see the commit [here](https://github.com/joaoantoniocardoso/kicad-exports/commit/5c9c5b3ebe767019d29a8d5e9fce1e178f900a20)) and here you are: [:1st_place_medal:  it is working!!](https://github.com/ZeniteSolar/MCB19/actions/runs/198581902) 

I don't mind to use it in my current fork for now, but do you mind talking about your plans for the next release?

Thanks

nerdyscout, 2020-08-07:
interesting... I can not build the container on dockerhub... seems like the submodules do not get checked out recursively why the copy fails. so it's the exact same issue.
I am wondering why you can use joaoantoniocardoso/kicad-exports@develop, it should be the same.

nerdyscout, 2020-08-07:
i don't get how you were able to build the container... locally everything is fine, but on dockerhub it fails for me - most likely because the submodules are not initialized. just like your error above, copying empty folders crashes of course.

joaoantoniocardoso, 2020-08-07:
> i don't get how you were able to build the container... locally everything is fine, but on dockerhub it fails for me - most likely because the submodules are not initialized. just like your error above, copying empty folders crashes of course.

Hmm, are you sure you are tagging and pushing the last image to dockerhub? In my case I am doing the following:
```bash
git clone --recursive https://github.com/nerdyscout/kicad-exports && cd kicad-exports
git checkout develop
git submodule update --init
make
make install >> install.sh
chmod +x install.sh
./install.sh
rm install.sh
docker tag kicad-exports:latest joaoantoniocardoso/kicad-exports:latest
docker push joaoantoniocardoso/kicad-exports:latest
```

nerdyscout, 2020-08-08:
I made it :fireworks: still don't know what went wrong but it seems fine now.

I tagged now a version as v2.0 and try several of my PCBs with it. So far I experienced your initial issue with the timeout once, besides that it seems to be good.

joaoantoniocardoso, 2020-08-09:
Yesterday I tested a project using v2.0 and it didn't went very well. Today I noticed some new commits, then I re-run now and everything seems to be working as expected!

joaoantoniocardoso, 2020-08-19:
I can't reproduce the problem anymore, so I am closing this.

closed 2020-08-19 by joaoantoniocardoso