nerdyscout/kicad-exports #13: Github Action can't be pulled

tabeatheunicorn, 2020-08-07:
I want to try your project as it looks REALLY nice. Took one of your example files and wanted to see how far I could get. 

So the Github Action doens't suceed in pulling kicad-exports.

` Pull nerdyscout/kicad-exports:v1.117s
  ##[error]Docker pull failed with exit code 1
Pull down action image 'nerdyscout/kicad-exports:v1.1'
  /usr/bin/docker pull nerdyscout/kicad-exports:v1.1
  Error response from daemon: manifest for nerdyscout/kicad-exports:v1.1 not found: manifest unknown: manifest unknown
  ##[warning]Docker pull failed with exit code 1, back off 5.682 seconds before retry.
  /usr/bin/docker pull nerdyscout/kicad-exports:v1.1
  Error response from daemon: manifest for nerdyscout/kicad-exports:v1.1 not found: manifest unknown: manifest unknown
  ##[warning]Docker pull failed with exit code 1, back off 7.518 seconds before retry.
  /usr/bin/docker pull nerdyscout/kicad-exports:v1.1
  Error response from daemon: manifest for nerdyscout/kicad-exports:v1.1 not found: manifest unknown: manifest unknown
  ##[endgroup
  ##[error]Docker pull failed with exit code 1`

I also tried @v1.0, @v2.0 and no version at all and none of them works.
If you have an idea on what I did wrong, I would really appreciate it.

`name: test

on:
  push:
    branches:
      - master
      - develop
  pull_request:
    branches:
      - master
      - develop

jobs:
  schematic:
    runs-on: ubuntu-latest
    steps:
    - uses: actions/checkout@v2
    - uses: nerdyscout/kicad-exports@v1.1
      with:
        cmd: schematic
        dir: test/docs
        board: test/test.kicad_pcb
        schematic: test/test.sch
    - name: kicad-schematic
      uses: andstor/file-existence-action@v1
      with:
        files: "test/docs/test_schematic.svg, test/docs/test_schematic.pdf"
    - uses: actions/upload-artifact@v2
      if: success()
      with:
        name: schematic
        path: test/docs/*
`

This is the .yml file I use

nerdyscout, 2020-08-07:
most likely the old versions do not work (anymore) as i just deleted and renewed the dockerhub access.
Why the upcoming v2.0 is not working is what i am currently looking for by myself... some issue with the submodules not been pulled from docker... don't know yet. please be patient.

nerdyscout, 2020-08-08:
there is now a version tagged with v2.0 which is still pre-relase, if you want you could try this one.

nerdyscout, 2020-08-21:
closing as this should be no issue anymore with the released v2.0

closed 2020-08-21 by nerdyscout