nerdyscout/kicad-exports #15: [BUG] No files are generated

Informaticore, 2020-09-29:
**Describe the bug**
I copy+pasted the action into my Git-repo and I try to run the docs.kiplot.yaml. It runs DRC and ERC and also prints its results into the files but the pdf and svg are not generated.
**To Reproduce**
Steps to reproduce the behavior:
You can find the dedicated branch here: https://github.com/Informaticore/SuperPower/tree/testing_github_action

**Expected behavior**
After running the action it uploads pdf and svg files as zip into the artifacts
**Additional context**
It seems that it is not really running the generation part, only the checks for DRC and ERC

nerdyscout, 2020-09-29:
I can see several commits running the action and generating all files like expected.
eg [117d10a](https://github.com/Informaticore/SuperPower/commit/117d1a04fb28ff73a9c338a6845785d56ff9fb07) seems fine. 
Do you need any more support?

Informaticore, 2020-09-29:
hey - sorry I just figured out afterwards, that I have to give the explicite path to the .sch and pcb file. Is there a way to make it every file in the project no matter where it is? I tried `**/*.sch` but it was not working.
Should it generate the board even if it is empty? Because it can not find the layers (but as I said the pcb file is still empty currently.

nerdyscout, 2020-09-29:
as you have several *.sch files you have to specify which one to use. otherwise it should pick 'alphabetical first .sch found in current dir' but there is still a little bug I was not able to get rid off... and you are using a subdir anyway
If you specify your main schematic file all subsheets should be created.

I have never tried it with an empty board file before, it fails for me too. 
If you add at least the Edge cuts it works.

Informaticore, 2020-09-29:
perfect! thank you! I will try it out later - very nice github action anyway! Exactly what we are looking for.
Thanks a lot.

closed 2020-09-29 by Informaticore