nerdyscout/kicad-exports #16: [BUG] Can't create BoM

tabeatheunicorn, 2020-10-01:
**Describe the bug**
I can't generate any BoM output, it fails.

**To Reproduce**
Steps to reproduce the behavior:
- used  this bom.kiplot.yaml file
`kiplot:
  version: 1
  
preflight:
  run_erc: true
  run_drc: true
  check_zone_fills: true
  ignore_unconnected: false
  update_xml: true
  filters:
    - filter: 'Ignore this'
      number: 45
      regex: 'Footprint U1'
    
global:
  output: '%f_%D.%x'
  
outputs:
  #- name: bom_html
   # comment: "Bill of Materials in HTML format"
    #type: kibom
    #dir: BoM
    #options:
     # format: HTML  # HTML or CSV
      #variant: pp

  - name: bom_csv
    comment: "Bill of Materials in CSV format"
    type: kibom
    dir: BoM
    options:
      format: CSV  # HTML or CSV
      conf:
        ignore_dnf: true
© 2020 GitHub, Inc.`

**Expected behavior**
I expect the program to generate a BoM and upload it to BoM
- Updating BoM in XML format
- 'Bill of Materials in CSV format' (bom_csv) [kibom]
ERROR:Failed to create BoM, error 1 (kiplot.kiplot.out_kibom - out_kibom.py:350)
 I don't know what this line states

nerdyscout, 2020-10-01:
if running locally please add `-v` and provide output.
if running somewhere on github tell me which repository so I could there.
kicad project data might be helpful as well.

tabeatheunicorn, 2020-10-05:
I have a repository containing a kicad project that I want to generate a bom from (both the interactive html and a csv version)
Therefore I want to use your action with a bom.kiplot.yaml file. Not sure how to format that file so that it is useful. 

What is probably interesting is the Error:
Bill of Materials in CSV format' (bom_csv) [kibom]
ERROR:Failed to create BoM, error 1 (kiplot.kiplot.out_kibom - out_kibom.py:350)

So it doesn't generate a bom but I don't know where to find out_kibom.py or what exactly the error is.

nerdyscout, 2020-10-05:
As your config does not pass any kind of lint test it is save to say your config is malformed, it is not an issue with the out_kibom.py

Be careful coping, as your initial config does include the string '© 2020 GitHub, Inc.' is is obvious something went wrong the last time. Also be aware in [yaml](https://en.wikipedia.org/wiki/YAML#Syntax) files every indentation matters!

Please try this config instead:
```
kiplot:
  version: 1

preflight:
  update_xml: true

outputs:
  - name: 'ibom'
    comment: "Interactive Bill of Materials (HTML)"
    type: ibom
    dir: docs/bom
    options:
      dark_mode: false
      hide_pads: true
      show_fabrication: true
      hide_silkscreen: true
      highlight_pin1: true
      no_redraw_on_drag: true
      board_rotation: 0
      checkboxes: 'Sourced,Placed'
      include_tracks: true
      include_nets: true
      sort_order: 'C,R,L,D,U,Y,X,F,SW,A,~,HS,CNN,J,P,NT,MH'
      blacklist: 'DNF*'
      no_blacklist_virtual: true
      blacklist_empty_val: true
      extra_fields: 'EF'
      normalize_field_case: true
      dnp_field: 'DNP'
      name_format: ''
      output: '%f.%x'

  # KiBoM (KiCad Bill of Materials):
  # For more information: https://github.com/INTI-CMNB/KiBoM
  # This output is what you get from the 'Tools/Generate Bill of Materials' menu in eeschema.
  - name: 'kibom'
    comment: 'kibom csv'
    type: 'kibom'
    dir: 'docs/bom'
    options:
      # [string='HTML'] [HTML,CSV,XML,XLSX] format for the BoM
      format: 'CSV'
      # [number=1] Number of boards to build (components multiplier)
      number: 1
      # [string='%f-%i.%x'] filename for the output (%i=bom). Affected by global options
      output: '%f.%x'
      # [string=','] CSV Separator
      separator: ';'
      conf:
        hide_pcb_info: false
        exclude_any:
          - column: Part
            regex: TestPoint
          - column: Part
            regex: LOGO
          - column: Part
            regex: Jumper
```

nerdyscout, 2020-10-05:
> As your config does not pass any kind of lint test it is save to say your config is malformed, so far it is not an issue with the out_kibom.py
ohhh... the weird indentation does occur because of the single backtick where three of them have to be used for multi line code blocks.
could you just give me access to your repo?

nerdyscout, 2020-10-24:
hi @tabeatheunicorn 
I really want to help you... could you please give me a link to the repository you are using it?
[bom](https://github.com/nerdyscout/kicad-exports/actions/runs/325773618) generation currently runs fine for me, but I might have a idea what is happening to you... i just want to double check.

tabeatheunicorn, 2020-10-30:
Hey, I can't give you access to that repo and didn't have time to set up anotherone accordingly. Will do so hopefully within the next few days.

tabeatheunicorn, 2020-11-02:
@nerdyscout I send you an invite to a repo which I can show to you. Thanks for helping out!

tabeatheunicorn, 2020-11-02:
Update on this: as you might notice, the generation of the bom file suceeds in my own repository (probably need to check some writing permissions?) 
Expected it to put the file in the doku folder though.

closed 2020-11-20 by nerdyscout