nerdyscout/kicad-exports #17: [BUG] Permission denied: '/github/home/.config'

joaoantoniocardoso, 2020-10-21:
**Describe the bug**
After upgrading from 2.0 to 2.1, my workflow starts to fail with the following error (see the CI [here](https://github.com/ZeniteSolar/MCB19/runs/1280901377?check_suite_focus=true#step:4:90)):
```PowerShell
PermissionError: [Errno 13] Permission denied: '/github/home/.config'
DEBUG:Checking if /github/workspace/hardware/MCB19.pro was modified (eeschema_do.kiauto.file_util - file_util.py:241)
ERROR:ERC returned 1 (kibot.kibot.pre_erc - pre_erc.py:45)
```

**To Reproduce**
Steps to reproduce the behavior:
- [This](https://github.com/ZeniteSolar/MCB19/commit/56917f199afc0ef314f368c75055a4ef2152a35f) patch is the only modification necessary to reproduce, note that there are several customized yaml files and the error occurs in all of them.

**Expected behavior**
I just [reverted the patch](https://github.com/ZeniteSolar/MCB19/pull/20) and it is just behaves how I expected 2.1 would too.

nerdyscout, 2020-10-22:
i forked https://github.com/nerdyscout/MCB19/ and using `kicad-exports:develop` it seems currently fine. 

in the plot workflow is a error because you are using a own config file and I have moved the style from
`/opt/pcbdraw/pcbdraw/styles/jlcpcb-green-hasl.json` to
`/opt/pcbdraw/styles/jlcpcb-green-hasl.json` after changing in your config that should be fine.

I hope to find the bug in the user permission commit and re-add id some when on the weekend.

joaoantoniocardoso, 2020-10-23:
Nice, thanks for your efforts.

I added the filter config files and the patch to fix the path of the pcbdraw to test with 2.1, it keeps with the same permission error. See [here](https://github.com/ZeniteSolar/MCB19/pull/21/files).

By just looking the messages I don't understand exactly what is the error. I would like to help you but for now I'll not have the time to further investigate. 

Let me know if you find anything that may fix this bug or if you need something, thanks!

joaoantoniocardoso, 2020-10-23:
@nerdyscout Just to mention that your last release on v2.1 somehow solves this issue (see CI [here](https://github.com/ZeniteSolar/MCB19/pull/21/checks?check_run_id=1300475125)). Thanks!

joaoantoniocardoso, 2020-10-23:
Not enterely related, but you mentioned changing pcbdraw styles but (see [here](https://github.com/ZeniteSolar/MCB19/runs/1300488611#step:4:63)):
``` PowerShell
DEBUG:Executing: ['pcbdraw', '-s', '/opt/pcbdraw/styles/jlcpcb-green-hasl.json', '-l', '/opt/pcbdraw/lib', '-f', '', '--no-warn-back', '--dpi', '300', '/github/workspace/hardware/MCB19.kicad_pcb', '/github/workspace/hardware/docs/imgs/MCB19-top.svg'] (kibot.kibot.out_pcbdraw - out_pcbdraw.py:92)
ERROR:Failed to run pcbdraw, error 1 (kibot.kibot.out_pcbdraw - out_pcbdraw.py:96)
DEBUG:Output from command: Missing 'styles' /opt/pcbdraw/styles/jlcpcb-green-hasl.json
```

Could you check the right path?

nerdyscout, 2020-10-24:
Hi @joaoantoniocardoso 
I am sorry you are going through all these troubles because of my software.
These two line is how it should look like. I will update the documentation later.
```yaml
      style: '/usr/share/pcbdraw/styles/jlcpcb-green-hasl.json'
      libs: ['/usr/share/pcbdraw/footprints/KiCAD-base']
```

joaoantoniocardoso, 2020-10-29:
Hi, oh, don't feel bad for having bugs, that is normal, this software is really useful.
Thanks again, everything is working fine again =)

closed 2020-10-29 by joaoantoniocardoso