nerdyscout/kicad-exports #20: pushing test assets back to repo fails

nerdyscout, 2020-11-08:
maybe someone could help me with this one:
https://github.com/github-actions-x/commit/issues/22
https://github.com/nerdyscout/kicad-exports/runs/1369860587

I did set up more config files which create more output. For documentation propose I want these outputs be pushed back to the repository... but this pushing fails and I can currently not figure out why.

joaoantoniocardoso, 2020-11-08:
Can you try to `git restore .` before `git push`?

nerdyscout, 2020-11-08:
basically I spent yesterday the whole day testing various [actions](https://github.com/marketplace?query=commit) to commit and push files. But I was not able to make a single one of them work like expected. Adding some extra commands would require to write my own git handling action - which I would like to avoid.

joaoantoniocardoso, 2020-11-08:
Yeah, I see.. Well, maybe you can try to use [another](https://github.com/marketplace/actions/github-pushhttps://github.com/marketplace/actions/github-push) while waiting for the return from the developers.

nerdyscout, 2020-11-08:
turns out writing a own git-commit-handling is not so hard if you just fork one :partying_face:
https://github.com/nerdyscout/commit/blob/patch-1/entrypoint.py

closed 2020-11-15 by nerdyscout