nerdyscout/kicad-exports #22: PNG export, missing rsvg-convert

rbonghi, 2021-03-02:
**Describe the bug**
When selected png format the kicad export fail with a error where is missing `rsvg-convert`
> WARNING:(W017) `rsvg-convert` not installed, using unreliable PNG/JPG conversion (kibot.kibot.out_pcbdraw - log.py:82)
> WARNING:(W018) If you experiment problems install `librsvg2-bin` or equivalent (kibot.kibot.out_pcbdraw - log.py:82)
> ERROR:Failed to run pcbdraw, error 1 (kibot.kibot.out_pcbdraw - out_pcbdraw.py:96)

**To Reproduce**

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

nerdyscout, 2021-03-02:
if I get [this](https://github.com/INTI-CMNB/KiBot/search?q=rsvg-convert&type=commits) right your issue is not about rsvg-convert.
Actually your log shows the TOP png is generated, but the BOTTOM fails.
Running you config and kicad files locally everything seems fine.
this will need some further investigations.

rbonghi, 2021-03-02:
Hi @nerdyscout,
Meanwhile, during this day, I changed my config.kibot.yml and I made my workaround.

Thank you for your support, It's a great tool!

ckuhtz, 2021-04-03:
what did you change?

closed 2021-04-25 by nerdyscout