nerdyscout/kicad-exports #23: [BUG] schema: and pcb: not actually optional?

ckuhtz, 2021-04-03:
**Describe the bug**
no matter what I put in

      # optional - schematic file
        schema: '*.sch'
      # optional - PCB design file
        board: '*.kicad_pcb'

the workflow complains about missing files. -b and -e are always specified.  yet, it says in the example they're optional.

**To Reproduce**
example kibot file in README.md
https://github.com/ckuhtz/cm4cluster (private at the moment)
Here's the excerpt from workflow run when both params are being omitted.
```
Run nerdyscout/kicad-exports@v2.2
/usr/bin/docker run --name nerdyscoutkicadexportsv22_2501e4 --label 5588e4 --workdir /github/workspace --rm -e INPUT_CONFIG -e INPUT_DIR -e INPUT_BOARD -e INPUT_SCHEMA -e HOME -e GITHUB_JOB -e GITHUB_REF -e GITHUB_SHA -e GITHUB_REPOSITORY -e GITHUB_REPOSITORY_OWNER -e GITHUB_RUN_ID -e GITHUB_RUN_NUMBER -e GITHUB_RETENTION_DAYS -e GITHUB_ACTOR -e GITHUB_WORKFLOW -e GITHUB_HEAD_REF -e GITHUB_BASE_REF -e GITHUB_EVENT_NAME -e GITHUB_SERVER_URL -e GITHUB_API_URL -e GITHUB_GRAPHQL_URL -e GITHUB_WORKSPACE -e GITHUB_ACTION -e GITHUB_EVENT_PATH -e GITHUB_ACTION_REPOSITORY -e GITHUB_ACTION_REF -e GITHUB_PATH -e GITHUB_ENV -e RUNNER_OS -e RUNNER_TOOL_CACHE -e RUNNER_TEMP -e RUNNER_WORKSPACE -e ACTIONS_RUNTIME_URL -e ACTIONS_RUNTIME_TOKEN -e ACTIONS_CACHE_URL -e GITHUB_ACTIONS=true -e CI=true -v "/var/run/docker.sock":"/var/run/docker.sock" -v "/home/runner/work/_temp/_github_home":"/github/home" -v "/home/runner/work/_temp/_github_workflow":"/github/workflow" -v "/home/runner/work/_temp/_runner_file_commands":"/github/file_commands" -v "/home/runner/work/cm4cluster/cm4cluster":"/github/workspace" nerdyscout/kicad-exports:v2.2  "-c cm4cluster.kibot.yaml" "-d docs" "-b " "-e "
error: -e does not exist
```

**Expected behavior**
optional should be optional

**Additional context**
coming new into this workflow, there really should be some description somewhere of how kicad-exports and kibot work together.  it's not clear to me how to make this work. :(

nerdyscout, 2021-04-03:
could you please add me to your private repo for further analysis?

nerdyscout, 2021-04-08:
@ckuhtz although not yet released please retry with current v2.3 / master as this should make optional parameters more optional.

nerdyscout, 2021-04-25:
v2.3 is now released.
without access to your repository I was never able to reproduce this behavior.

closed 2021-04-25 by nerdyscout