nerdyscout/kicad-exports #24: [FEATURE REQUEST] Generate ".brd" file

joaoantoniocardoso, 2021-04-18:
Hi!

Would it be possible to integrate one more tool into this toolbox? I think exporting ".brd" files would be very interesting. Quoting directly from [this tool](https://github.com/whitequark/kicad-boardview) which I am currently using:

> The KiCAD to Boardview exporter reads KiCAD PCB layout files and writes Boardview files. Boardview is a set of file formats used in the industry for communicating important data about PCB in post-production stage, such as component, pin, and test point placement.
> 
> The Boardview file formats do not typically have a specification or even a name. The KiCAD to Boardview exporter writes ".brd" files with ASCII Boardview data. These files can be opened with MIT-licensed OpenBoardView.

Sincerely I don't really know how many people would be benefited from it but I think it would be great to be able to automatically generate the board view right from the github actions.

Let me know what you think about it,
Thanks!

nerdyscout, 2021-04-18:
hi @joaoantoniocardoso,
I vaguely remember someone already had the same wish... at least I already put this tool on the [wish list](https://github.com/nerdyscout/kicad-exports/wiki#tools-i-would-love-to-see-integrated-into-kibot).
Technically it should be no big deal to integrate, but I would prefer to see it integrated into kibot.

joaoantoniocardoso, 2021-04-19:
Hi @nerdyscout, nice!

I couldn't find any mention about .brd files there, have you already asked this at [kibot](https://github.com/INTI-CMNB/KiBot/issues)?

Thanks!

nerdyscout, 2021-04-19:
no I did not... I prefer creating pull requests instead of demanding for features, but just don't have the time and knowledge. especially because there are more tools (kikit) I would like to see integrated.

joaoantoniocardoso, 2021-04-19:
hmm, as a developer I would prefer to see issues first to better discuss the subject matter, maybe it is the case that someone is already working on that or at least have something to share, like that unexpected golden tip from other previous implementations.

I think I'll create it there and then link it back here, thanks!

nerdyscout, 2021-04-19:
> hmm, as a developer I would prefer to see issues first to better discuss the subject matter, maybe it is the case that someone is already working on that or at least have something to share, like that unexpected golden tip from other previous implementations.

fair point.

joaoantoniocardoso, 2021-04-21:
Hi @nerdyscout, I am just coming back here with good news - KiBot now supports .brd file generation on master (see the merged commits [here](https://github.com/INTI-CMNB/KiBot/commit/cfe61c9b7e940eadc86bd099cb0a2b4055980e2b))!

I tested locally and seems very straightforward to integrate here since there are not many options available, what do you think?

nerdyscout, 2021-04-21:
thank you for pushing this topic. I will include it in the next release

nerdyscout, 2021-04-25:
v2.3 does have a sample config to generate boardview files
https://github.com/nerdyscout/kicad-exports/blob/v2.3/config/boardview.kibot.yaml

closed 2021-04-25 by nerdyscout