nerdyscout/kicad-exports #26: [BUG] Permission denied and fails to run ERC

yanghu, 2021-05-05:
**Describe the bug**
When trying to run kicad export to generate documentation(pdf, svgs). I got the following error 
the preflight erc check fails to run due to permission error.

```
- Running the ERC                                                                                                                                                                            DEBUG:Executing: ['eeschema_do', '-r', '-v', 'run_erc', '-o', '/home/huyang/workdir/unicorn-erc.txt', '/home/huyang/workdir/unicorn.sch', '/home/huyang/workdir/.'] (kibot.kibot.kiplot - kiplot.py:164)                                                                                                                                                                                  DEBUG:Failed with error 1, retrying ... (kibot.kibot.kiplot - kiplot.py:171)                                                                                                                 WARNING:(W058) Missing KiCad main config file /home/huyang/.config/kicad/kicad_common (eeschema_do - misc.py:161) (kibot.kibot.kiplot - log.py:82)                                           DEBUG:Output from command:                                                                                                                                                                   > WARNING:Missing KiCad main config file /home/huyang/.config/kicad/kicad_common (eeschema_do - misc.py:161)                                                                                 > Traceback (most recent call last):
>   File "/usr/bin/eeschema_do", line 561, in <module>                                                                                                                                       >     create_eeschema_config(cfg)                                                                                                                                                            >   File "/usr/bin/eeschema_do", line 446, in create_eeschema_config                                                                                                                         >     with open(cfg.conf_eeschema, "wt") as text_file:                                                                                                                                       > PermissionError: [Errno 13] Permission denied: '/home/huyang/.config/kicad/eeschema'                                                                                                       >  (kibot.kibot.kiplot - kiplot.py:175)
ERROR:ERC returned 1 (kibot.kibot.pre_erc - pre_erc.py:65) 
```

**To Reproduce**
Steps to reproduce the behavior:
install kicad-exports locally with docker, then run this on my project.
`kicad-exports -c /opt/kibot/config/bundle/documentation.default.kibot.lst -v`

The schema has no ERC error when I run erc manually in eeschema on my local machine.

**Expected behavior**
erc should have no error and produce the output files.

**Additional context**
Add any other context about the problem here.

yanghu, 2021-05-06:
I was able to fix this issue by manually changing the owner and group of `.config/kicad`, as well as `.cache/kicad` ( for eeschema lock file checking during preflight erc)

I can get past erc/drc if I run the kicad-export command at the root folder of my project, and using `-e` flag to specify the schema file. 

the folder structure is like this

```
root
|--sym (contains symbols)
|--design (contains kicad files)
```

However, if I run the command in the design subfolder where sch file is, I'll get `eeschema timeout` error for erc, I tried `-v` to debug the issue, and the problem is the symbol libraries fails to load.

I added some libraries using the relative path like this
`${KIPRJMOD}/../../sym/foo/Connector.lib`

If I run the command at root, where "sym" is also there, the erc check would pass.

Is there anything I can do to be able to run the `kicad-export` command at the designs folder?

nerdyscout, 2021-05-06:
Hi @yanghu,
````
docker run --volume="$PWD:/home/user/workdir:rw" \
    --volume="/etc/group:/etc/group:ro" \
    --volume="/etc/passwd:/etc/passwd:ro" \
    --volume="/etc/shadow:/etc/shadow:ro" \
    --volume="/tmp/.X11-unix:/tmp/.X11-unix" \
    --volume="/home/user/.config/kicad:/home/user/.config/kicad:rw" \
    --volume="/home/user/.cache/kicad:/home/user/.cache/kicad:rw" \
    --workdir="/home/user/workdir" \
    --user $(id -u):$(id -g) \
    kicad-exports $@
````

`./config/kicad` and `./cache/kicad` are mapped from your user into the docker container. currently I don't know why you need to change user permission, processes inside docker should run with the same user as you are... this is something I will try to fix.
 
`kicad-exports` maps the folder from where it is started (pwd) to `workdir` inside the docker container. you won't be able to reach your symbols. sorry but this is intended like this. you already found the best solution for this, just use the `-e `parameter.

yanghu, 2021-05-06:
Thanks for the quick response! I understand that there's limitation on mapping folders to docker. I was wondering if there's some workarounds to allow us map a parent folder to docker instead of pwd. but I understand that it might not worth it.

closed 2021-05-06 by yanghu