nerdyscout/kicad-exports #27: [Feature request] Apply/override preflight filters to all configs in a bundle

yanghu, 2021-05-06:
this is not a bug, but a feature request.

In my workflow, I want to run same set of configs(yaml files) on my PCB designs. 

Say, I have workflow X, Y and Z that I want to run every time. (generate gerber, bom, position files)

I also have Kicad projects A, B and C. 

However, I do want to have some specific drc errors ignored for my projects. Issue is that, the filter would be project-specific. But the preflight filters are hard-coded inside X,Y and Z, meaning it would be the same filters for all projects.

Now my solution is to fork X,Y,Z for projects, so I have Xa, Ya, Za with filters for project A, and run it for project A. same for B and C. This is obviously not ideal

My goal is to reuse the same set of yaml files, but when running kicad-exports, I can run preflight with project specific filters.

One way is to remove all preflight in my yamls, and for each project, I write a small bundle file. Question is, can I add preflight to bundle files? Seems like bundle is a feature introduced in kicad-exports (not from kibot).  If we can specify preflight in bundles, I can simply write a bundle for each project A, B and C, reusing the same set of yamls(XYZ), but with different preflight filters in the bundle files.

nerdyscout, 2021-05-06:
Hi again :)

yes bundles are introduced by kicad-exports, not kibot. As bundles are just a list of kibot.yaml files there is no way to skip the preflight there. But running in the terminal you could add the parameter `-s all` which will skip all preflights. Honestly I have never tested this feature with bundles but I can not see why this should not work.

If I get your case correct I would try to generate a filter1.yaml for project1, filter2.yaml for project2, ...  and a generic documentation.yaml, gerber.yaml, ... for different output files common to all projects. Then you could bundle one filter and all generic configs to end up with bundle1.lst, bundle2.lst.

If you find some good generic configs I would be happy to integrate them into kicad-exports!

yanghu, 2021-05-06:
Yes, `-s all` would work. (I'm currently not using bundle, but using multiple `-c` flags in a makefile.

However, I still would like to run drc, but just with filters to skip specific errors like pads too close for a certain part in a certain design. `-s all` would lose drc at all.

A separater filter.yaml for a project sounds good, but, how does it work?  if the separate filter.yaml runs preflight check, a drc failure would not prevent subsequent yamls in the bundle to run, so we need to manually check for drc failures.(and we'll get wrong artifacts generated anyway)

Note that we won't run preflight drc in generic yaml like "gerber.yaml" because we can't add filters there. So the only yaml runs drc would be the `filter1.yaml`.  Will a failure there abort the whole bundle run?

Or is it possible to access filters or preflight options defined in my `filter1.yaml` from the generic workflow yaml?  

I've tried the default bundles shipped with kicad-export, what I saw is that, each yaml has preflight check option defined, and one yaml failing preflight check won't stpo the other yamls to run. but maybe I'm misunderstanding how preflight checks work and how bundles work with preflight checks.

nerdyscout, 2021-05-06:
> Yes, `-s all` would work. (I'm currently not using bundle, but using multiple `-c` flags in a makefile.
> 
> However, I still would like to run drc, but just with filters to skip specific errors like pads too close for a certain part in a certain design. `-s all` would lose drc at all.

Besides `-s all` there are is `-s xxx` xxx= more options provided by kibot. I just never used them and don't remember. 
 
> A separater filter.yaml for a project sounds good, but, how does it work? if the separate filter.yaml runs preflight check, a drc failure would not prevent subsequent yamls in the bundle to run, so we need to manually check for drc failures.(and we'll get wrong artifacts generated anyway)

You could try to catch the return code of kicad-exports running the preflight.yaml and run the further *.yaml only if passed.

> Note that we won't run preflight drc in generic yaml like "gerber.yaml" because we can't add filters there. So the only yaml runs drc would be the `filter1.yaml`. Will a failure there abort the [...]

[importing-outputs-from-another-file](https://github.com/INTI-CMNB/KiBot#importing-outputs-from-another-file)
This is a feature I just learned about! (otherwise I wouldn't have implemented bundles) this sounds like it would fit your needs.

> I've tried the default bundles shipped with kicad-export, what I saw is that, each yaml has preflight check option defined, and one yaml failing preflight check won't stpo the other yamls to run. but maybe I'm misunderstanding how preflight checks work and how bundles work with preflight checks.

yes the default configs all run preflights as I wanted to avoid users being able to export faulty data, and blaming me for ending up with bad PCBs. 
At the same time it forces lazy users (like me) to design proper PCBs without any ERC/DRC violations.

yanghu, 2021-05-06:
"You could try to catch the return code of kicad-exports running the preflight.yaml and run the further *.yaml only if passed."
Are you saying doing this in the script(outside kicad-exports), like in a makefile? sounds like a good idea. The solution doesn't have to be done inside `kicad-export`.
(it would be great to support such behabior in bundles though, for example, stop when one of the yaml fails, instead of continue)

I tried importing outputs, but seems like it can only import `output` section, not other stuff like variants and filters. I got error when imported [this file](https://github.com/INTI-CMNB/KiBot/blob/master/docs/samples/JLCPCB_position.kibot.yaml) with error saying `rotated` is not defined, which is a variant defined in that file. 

Regarding drc violations, the ones I'm getting are "pad to close to pad" which is intentional because I was stacking two footprints in one place to allow alternative use of the same space, (it would also cause courtyard collision, if courtyard is properly defined for each footprint). In such cases, filters are a must. (or manually run DRC before executing the export command). 

Thanks for your reply! I'll take your suggestion, run two separate `kicad-export` commands, first one to do DRC/ERC, and second one to generate artifacts without checking DRC/ERC. the failure of first execution would abort the operation.

However, native support of such behavior in bundles would simplify the workflow. (for example, github actions and local execution requires different setup to do this). Do you think it's worth to add the feature?

nerdyscout, 2021-05-06:
personally I would prefer to keep as much as possible of all functionality inside kibot. kicad-exports is meant to be full environment, mainly for the github CI, to not care about dependencies. I think, besides me running it a lot local as well, you are the first using it in a similar way :)
Building advanced logic into kicad-exports might be interfere with future version of kibot.

afaik KiCad6 will bring some breaking changes to ERC/DRC. Therefore I would rather wait till it is released and built on that.

yanghu, 2021-05-06:
Thank you! I initially filed a similar bug in kibot but then realized bundles is not introduced there, then filed this one here. 
I think something like a hand-written "kibot.drc.filter" file in the design project folder would work really well for this case. 
I know kibot automatically creates one, but if it can load it from the design folder, and not overwrite it, that would be all I need.

I'm new to kibot so not sure if such feature is already doable.

by the way, what's your suggestion for my initial question? any trick to do this inside kibot? Or do you think running separate commands for drc and artifacts generation is the way to go? (I beliebe github actions also fails the whole action if one step fails, which is what I want here)

closed 2021-05-11 by nerdyscout

nerdyscout, 2021-08-28:
hi @ModischFabrications 
setting up bundles in the CI is [as easy as this](https://github.com/nerdyscout/turn_indicator_pcb/blob/bundleaction/.github/workflows/exports.yml). just be aware you need to use an absolute path for the bundle. (seems like a bug to me, didn't check why so far).

Personally I use bundles in local mode to have very quick everything together needed to upload to the PCB manufacturer.
If you do bundles in the CI you will need to keep the generated files somewhere external (cloud/ftp/...) as you already noticed the limited retention.