nerdyscout/kicad-exports #28: [BUG] version 2.2 doesn't support "rot_footprint" and 2.3 doesn't run at all

thejpster, 2021-06-08:
**Describe the bug**
I am using this github action to build my Kicad 5 project. It has been working great, but I had to add a footprint rotation to fix a JLCPCB issue. If I use v2.2, it says 

```
Run nerdyscout/kicad-exports@v2.2
  with:
    config: Kicad/docs.yml
    dir: docs
    schema: Kicad/neotron-pico.sch
    board: Kicad/neotron-pico.kicad_pcb
/usr/bin/docker run --name nerdyscoutkicadexportsv22_b62873 --label 8a33c1 --workdir /github/workspace --rm -e INPUT_CONFIG -e INPUT_DIR -e INPUT_SCHEMA -e INPUT_BOARD -e HOME -e GITHUB_JOB -e GITHUB_REF -e GITHUB_SHA -e GITHUB_REPOSITORY -e GITHUB_REPOSITORY_OWNER -e GITHUB_RUN_ID -e GITHUB_RUN_NUMBER -e GITHUB_RETENTION_DAYS -e GITHUB_ACTOR -e GITHUB_WORKFLOW -e GITHUB_HEAD_REF -e GITHUB_BASE_REF -e GITHUB_EVENT_NAME -e GITHUB_SERVER_URL -e GITHUB_API_URL -e GITHUB_GRAPHQL_URL -e GITHUB_WORKSPACE -e GITHUB_ACTION -e GITHUB_EVENT_PATH -e GITHUB_ACTION_REPOSITORY -e GITHUB_ACTION_REF -e GITHUB_PATH -e GITHUB_ENV -e RUNNER_OS -e RUNNER_TOOL_CACHE -e RUNNER_TEMP -e RUNNER_WORKSPACE -e ACTIONS_RUNTIME_URL -e ACTIONS_RUNTIME_TOKEN -e ACTIONS_CACHE_URL -e GITHUB_ACTIONS=true -e CI=true -v "/var/run/docker.sock":"/var/run/docker.sock" -v "/home/runner/work/_temp/_github_home":"/github/home" -v "/home/runner/work/_temp/_github_workflow":"/github/workflow" -v "/home/runner/work/_temp/_runner_file_commands":"/github/file_commands" -v "/home/runner/work/Neotron-Pico/Neotron-Pico":"/github/workspace" nerdyscout/kicad-exports:v2.2  "-c Kicad/docs.yml" "-d docs" "-b Kicad/neotron-pico.kicad_pcb" "-e Kicad/neotron-pico.sch"
ERROR:Unknown filter type: `rot_footprint` (kibot.kibot.error - error.py:40)
```

If I upgrade 2.3, it doesn't produce any outputs.

```
Run nerdyscout/kicad-exports@v2.3
  with:
    config: Kicad/docs.yml
    dir: docs
    schema: Kicad/neotron-pico.sch
    board: Kicad/neotron-pico.kicad_pcb
/usr/bin/docker run --name a33c1a0c898db0a2f479dadee8b30b0a17b4e_d4af69 --label 8a33c1 --workdir /github/workspace --rm -e INPUT_CONFIG -e INPUT_DIR -e INPUT_SCHEMA -e INPUT_BOARD -e INPUT_VERBOSE -e HOME -e GITHUB_JOB -e GITHUB_REF -e GITHUB_SHA -e GITHUB_REPOSITORY -e GITHUB_REPOSITORY_OWNER -e GITHUB_RUN_ID -e GITHUB_RUN_NUMBER -e GITHUB_RETENTION_DAYS -e GITHUB_ACTOR -e GITHUB_WORKFLOW -e GITHUB_HEAD_REF -e GITHUB_BASE_REF -e GITHUB_EVENT_NAME -e GITHUB_SERVER_URL -e GITHUB_API_URL -e GITHUB_GRAPHQL_URL -e GITHUB_WORKSPACE -e GITHUB_ACTION -e GITHUB_EVENT_PATH -e GITHUB_ACTION_REPOSITORY -e GITHUB_ACTION_REF -e GITHUB_PATH -e GITHUB_ENV -e RUNNER_OS -e RUNNER_TOOL_CACHE -e RUNNER_TEMP -e RUNNER_WORKSPACE -e ACTIONS_RUNTIME_URL -e ACTIONS_RUNTIME_TOKEN -e ACTIONS_CACHE_URL -e GITHUB_ACTIONS=true -e CI=true -v "/var/run/docker.sock":"/var/run/docker.sock" -v "/home/runner/work/_temp/_github_home":"/github/home" -v "/home/runner/work/_temp/_github_workflow":"/github/workflow" -v "/home/runner/work/_temp/_runner_file_commands":"/github/file_commands" -v "/home/runner/work/Neotron-Pico/Neotron-Pico":"/github/workspace" 8a33c1:a0c898db0a2f479dadee8b30b0a17b4e  "-c Kicad/docs.yml" "-d docs" "-b Kicad/neotron-pico.kicad_pcb" "-e Kicad/neotron-pico.sch" "-v "
```

**To Reproduce**
- https://github.com/Neotron-Compute/Neotron-Pico, on the "add-test-pads" branch
- the kibot file is "Kicad/docs.yml"

**Expected behavior**

To get PDFs, gerbers, BoM and PnP files.

nerdyscout, 2021-06-08:
Hi @thejpster 

kicad-exports@v2.2 is based on kibot@v0.8.0, but rotation filters where introduced in v0.9.0. therefore this can not work.

for running with kicad-exports@v2.3 please change `config: Kicad/docs.yml` to `config: Kicad/docs.kibot.yaml` (and of course the filename ).
Actually I am not sure if *.yml files where ever accepted, if so it was not intended like this. it should be `*.kibot.yaml`

After changing it does produce outputs. I just had a quick look, seems like there are more issues. It will need a moment to sort out if this is related to your kicad-exports, kibot, or your project.

thejpster, 2021-06-08:
Thanks for the quick response. I have renamed the file, but I still see the same issue in 2.3 using a local Docker container.

```console
$ kicad-exports -d github_docs -e neotron-pico.sch -b neotron-pico.kicad_pcb -c docs.kibot.yml
kicad-exports version: v2.3 - 2021-06-08
```

thejpster, 2021-06-08:
Adding `set -x` to the entrypoint.sh gives:

```
kicad-exports version: v2.3 - 2021-06-08
jonathan@90c96448f6cc:~/workdir$ ./entrypoint.sh  -c docs.kibot.yml 
+ ary=()
+ SCRIPT=kicad-exports
+ margs=1
+ CONFIG=
+ BOARD=
+ SCHEMA=
+ SKIP=
+ DIR=
+ OVERWRITE=
+ VERBOSE=
+ COMMIT=
+ '[' ']'
+ DISPLAY=:0
+ EXIT_ERROR=1
++ xargs
+ args='-c docs.kibot.yml'
+ IFS=' '
+ read -r -a args
+ main -c docs.kibot.yml
+ margs_precheck 2 -c
+ '[' 2 -lt 1 ']'
+ args_process -c docs.kibot.yml
+ '[' -n -c ']'
+ case "$1" in
+ shift
+ [[ docs.kibot.yml == *.kibot.yaml ]]
+ [[ docs.kibot.yml == *.kibot.lst ]]
+ shift
+ '[' -n '' ']'
+ msg_version
+ echo -e 'kicad-exports version: v2.3 - 2021-06-08'
kicad-exports version: v2.3 - 2021-06-08
+ run
+ '[' -d .git ']'
+ '[' ']'
+ '[' ']'
+ return 1
```

thejpster, 2021-06-08:
Oh, it's *yaml*

thejpster, 2021-06-08:
That's fixed it :)

closed 2021-06-08 by thejpster

nerdyscout, 2021-06-08:
yes, I am sorry. I will make the **a** optional in future.

btw I haven't seen such a nice clean schematic for a long time. great work!