nerdyscout/kicad-exports #30: [Bug] unwanted *.ogv files

ModischFabrications, 2021-08-11:
**Describe the bug**
I'm unsure if I misconfigured my action or it this is actually an error. 

I get pcbnew_run_drc_screencast.ogv and similar *.ogv files in every directory of my export, but I would like to skip them. 
Skipping their upload as artifact should probably be possible, but not generating them in the first place seems even better. 

**To Reproduce**
Steps to reproduce the behavior:
- used slightly modified docs.schema.kibot.yaml file
- https://github.com/ModischFabrications/turn_indicator_pcb/

**Expected behavior**
I expected to get only the files that I ordered. check*.txt are fine, even if not actually requested, but *.ogv seem useless to me. 

**Additional context**
Add any other context about the problem here.

nerdyscout, 2021-08-14:
Hi @ModischFabrications 
*.ogv files are only created in verbose mode. So setting ´verbose:0´ should solve this.

might be a good idea to set this default in my readme... I will check

ModischFabrications, 2021-08-14:
Oh alright, I thought that was just for logging purposes, didn't realize the pictures were a result of that. Are there any other consequences of this option? Thanks for looking into it!

nerdyscout, 2021-08-14:
I didn't check now but I guess there are a few logfiles not created / created with different depth of information according to verbosity level.

ModischFabrications, 2021-08-18:
Removing verbosity from files worked as well, which is even easier to explain. Issue can be closed from my side. 

Nice side-effect: My generated /docs zip is now 0.7Mb instead of 2.

closed 2021-08-19 by nerdyscout